Lucifer06/RemoteGPIO #1: install/uninstall issues

kwindrem, 2023-12-08:
I tried installing your package and failed because there is no file set for the firmware I'm currently running. That's fine, BUT when the install fails, the uninstall section of the script runs and throws errors:

++ Uninstalling RemoteGPIO
rm: can't remove '/etc/rcS.d/S90rgpio_pins.sh': No such file or directory
rm: can't remove '/dev/gpio/relay_3': No such file or directory
rm: can't remove '/dev/gpio/relay_4': No such file or directory
rm: can't remove '/dev/gpio/relay_5': No such file or directory
rm: can't remove '/dev/gpio/relay_6': No such file or directory

up to date, audited 72 packages in 4s

1 package is looking for funding
  run `npm fund` for details

5 vulnerabilities (3 moderate, 2 high)

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.

up to date, audited 72 packages in 3s

1 package is looking for funding
  run `npm fund` for details

5 vulnerabilities (3 moderate, 2 high)

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.

up to date, audited 72 packages in 3s

1 package is looking for funding
  run `npm fund` for details

5 vulnerabilities (3 moderate, 2 high)

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
rm: can't remove '/data/rgpio': No such file or directory
++ RemoteGPIO uninstalled

You can sidestep the rm errors by using the -f option

The npm commands appear to be doing a lot and taking time. You should test for a proper install prior to running those commands.

Your ReadMe says this will only work on the Large version of Venus OS, but there is no check for this in your setup script.  I also ran this on my Raspberry PI test system and if it's only for Cerbo, there should also be a check for that also.

Your ReadMe also says the instal may take several minutes. This could be problematic during reinstall following a firmware update. /data/SetupHelper/reinstallMods runs as a background task which calls your setup script. It calls the other package setup scripts as well. So your script could significantly delay reinstall of other packages, especially with a slow internet connection. Also SetupHelper does not check for network resources before calling setup scripts so this could block completion of your script and/or others called from reinstallMods.

My recommendation would be to move the node red installation out of the main setup script and find a way to run that independently somehow after you are sure the network is functioning.

Let me know if I can help in any way.

Lucifer06, 2023-12-08:
Hi Kevin,

OK thanks for the catch.
I will review and update accordingly.
Do you know a simple way to test this is running on a Cerbo?
I think I have an idea I need to test for speeding-up the Node-Red dependencies install.

Cheers,
FreD.

------------------------------------------------------------------------------
The water is always bluer on the other side of the ocean ;-)

kwindrem, 2023-12-08:
/etc/venus/machine contains text string that indicates the platform. SetupHelper initializes $machine to that value also.

machine			     GS Device
ccgx		CCGX
einstein	Cerbo GX
cerbosgx	Cerbo SGX
bealglebone	Venus GX
canvu500	CanVu 500
nanopi		Multi/Easy Solar GX
raspberrypi2	Raspberry Pi 2/3
raspberrypi4	Raspberry Pi 4
ekrano		Ekrano GX

—
Kevin

Lucifer06, 2023-12-13:
I have fixed the uninstall errors.
I'm also executing the setup only if Venus OS Large is installed.
Regarding the requirement to install only in GX, assuming I go with the post install setup on Node-RED GUI, it will be possible to use any Venus machine with the TCP Relay version.

Lucifer06, 2024-03-04:
This is now fixed with v3.0.x

closed 2024-03-04 by Lucifer06

kwindrem, 2024-03-15:
I forgot to mention I was able to allow SetupHelper to install without the automatic GUI restart.