Lucifer06/RemoteGPIO #5: version number not consistent with Victron version scheme

kwindrem, 2023-12-17:
I see you are using the traditional vX.Y.Z version format, however Victron is using vX.Y~Z where the ~Z portion is for release versions. In that scheme, v1.2 is interpreted as later than v1.2~5

Right now, SetupHelper will interpret v1.2.1 as later than v1.2.0 but v1.2 will be the latest of the three.

I also detect versions like v2.3d1, v2.3.a1 v2.3b1 as development, alpha test and beta test, respectively and rank them in that order: v3.2b1 is the "newest". But v2.3 would be newer than all the others.

How would you indicate a prerelease version number? A released version (which must be identified as newer than a release).

I'm not sure I can accommodate vX.Y.ZbW format and still support Victron versioning. Right now, the .W portion would be ignored.

Lucifer06, 2023-12-21:
Thx Kevin, good to know. I suppose you recommend we follow Victron's versioning rule?

kwindrem, 2023-12-21:
Yes, it would be easiest to follow their rules but with some limitations, I can handle some variation.

The main limitation is only three numbers for major, minor and prelease and only two major, minor for a released build. Within that, I can accommodate your format as long as I know what to expect.

Lucifer06, 2023-12-21:
OK I think I can accommodate to Victron rule. Excuse possible typo: swallowed by my iPhone

kwindrem, 2023-12-21:
If you want to stick with v1.2.3, that's fine as long as you keep releases greater than test releases. That works now.

Lucifer06, 2023-12-21:
👍Excuse possible typo: swallowed by my iPhone

closed 2024-03-04 by Lucifer06