Lucifer06/RemoteGPIO #9: Feature Suggestion v3.0beta number of relays per module

thomasinaz41213, 2024-02-29:
Just an observation, but I noticed in the code it seems you have hard coded the modules to 8 relays and 8 DI each for a total of 16 & 16 across 2 modules. Can you consider making it configurable in the settings for the number of relays per module? For example I have 2 modules of 4 relays and 4 DI each, and one module of 2 relays and 2 DI. There are a few others on the forum that have also mentioned using different sized modules.

Lucifer06, 2024-02-29:
Let me finish the first release as it is and I will add configuration of number of I/O per module. Should be straight forward.

Lucifer06, 2024-02-29:
So I guess the variants are 2, 4, 8 and 16 IO per module.

Lucifer06, 2024-02-29:
I see there is also a 32 IO variant...

thomasinaz41213, 2024-03-05:
Will you also consider making available more than 2 modules? I currently have 3 and some people may be interested in more. Being able to use WiFi and tcpip makes retrofitting existing boats/RVs much more attainable. Some installations would be a nightmare to try to run new wiring in certain locations.

thomasinaz41213, 2024-03-05:
Awesome work by the way. I have to wonder if you ever sleep.👍

Lucifer06, 2024-03-06:
:-)
The supports fo more than 2x units will require some changes from both sides: RemoteGPIO but also GuiMods.
Did yo test the current beta version with 2x units? Working fine?

thomasinaz41213, 2024-03-07:
I finally got around to testing it on my production system today. Install was successful and enabling 2 modules added the relays and DI. However I was not able to get the relays working. I am not using rs485 USB, I am using tcpip. I put in the correct IP address for both modules, but was not able to add the port number, they use port 80 for a web interface, but use 502 for tcpip coms. Also fyi, I have the modules connecting thru the Venus AP and not regular wifi connection. I found the wifi access to be very unstable when not going thru the Venus AP. I have been running them this way for several months with good results.

What port is the driver attempting to use for coms? It should be 502.

Lucifer06, 2024-03-07:
Hi, the port entry should be displayed only if the unit is set to be controlled in RS485. This is for selection wich USB port is used.
If you select TCP control then there should be no port request.  
Have a look at rgpuo_driver, I'm using modpoll with only Unit ID and IP adress. So it uses TCP port by default. I don't see the need to configure it in RemoteGPIO as default setting of Dingtian and modpoll is already 502. 
I will double-check my above claim after my morning coffee. Stay tuned...

Lucifer06, 2024-03-07:
Have you changed the protocol to TCP ?
[image]

Then the Port choice disappear from the GUI:
[image]

thomasinaz41213, 2024-03-07:
Yes I changed to TCP and entered the correct IP address. The port I was referring to is TCP port 502 which you said should be automatic for mod poll, so not an issue.

This is on my production system which has been using the relay modules on TCP via node red without any issues.

thomasinaz41213, 2024-03-07:
Figured it out. There was a conflict in the dingtian settings. It is working over TCP now.

Lucifer06, 2024-03-07:
:-)Excuse possible typo: swallowed by my iPhone

Lucifer06, 2024-03-10:
I'm thinking to modify the menus so you can configure the number of units, within each unit select number of I/Os , with the current limit of adding a total of 16x I/O, so for example:
4x units of 4x I/O
or 1x unit of 16x I/O
or one unit 8x I/O + 2x 4x I/Os
or any combinaisons.

If we have to go with more than 16x it will also require Kevin to update GuiMods. I'm afraid beyond 16x GUI becomes too slow. Anyway I think 16x is already a lot!
FYI I'm personally already using 16x with two 8x I/O.

thomasinaz41213, 2024-03-12:
In my system I have 3 dingtian modules. I am also using switches connected to digital inputs of the dingtian to manually open/close some of the dingtian relays for items such as a water pump switch located in the head and some light switches located throughout. Is it possible to also have the rgpio_driver read the relay status so that the gui is correctly updated when a relay is opened/closed by a source other than the gui? Currently I have to rely on node red to keep things in sync.

Lucifer06, 2024-03-12:
Hello Thomas, At this stage, I'm not reading the relay status as I went quite simple with no risk of dead loop between the status read and the request to change the state. I'll add this in the todo list as I agree this is desirable. So most probably the solution is when the D-Bus driver request to close/open a relay, I will not u^date its status until it reports the desired position. I'm happy to ear from you how you would handle it.

Lucifer06, 2024-03-31:
Thomas did a great job supporting 2, 4, 8 and 16 relai modules, as well as up to 3x modules.
Well-done!
This is integrated in version 3.2

closed 2024-03-31 by Lucifer06