Lucifer06/RemoteGPIO #14: install on non-large Venus OS fails with inappropriate exit code

kwindrem, 2024-03-07:
I tried to install RemoteGPIO on the normal (not large) Venus OS image for v3.22 and on a Raspberry PI 4. It fails and uninstalls.

ERROR: no path to /usr/lib/node_modules .....
Then a bunch of org.freedesktop.DBus.Error lines
then a kill: usage error twice
Then a bunch of org.freedesktop.DBus.Error lines
ln: /service/rgpio/rgpio: File exists
nohup: appending output to nohup.out
++ RemoteGPIO installed
++ Uninstalling RemoteGPIO

the strange thing is the exit code is 252 which is EXIT_FILE_SET_ERROR

You should probably test for the large version if you need it for your package. SetupHelper has no checks for large vs normal image. You can use com.victronenergy.platform/Services/NodeRed/Mode to test for node red. The parameter will not be valid if not installed. (I looked in PageSettings.qml for how they test for it.

Lucifer06, 2024-03-07:
good catch, I haven't yet tried  on non large, and I thought I removed all the dependencies. 
I will have a look...

Lucifer06, 2024-03-07:
On a Large Image, Node-Red could be disabled during the installation of RemoteGPIO, and enabling it later would not provide the resources for the additional DI and Relays.
I want to install the json regardless Node-Red is running or not. 
So I'm checking com.victronenergy.settings /Settings/System/ImageType instead of com.victronenergy.platform/Services/NodeRed/Mode.
I have also fixed all the reported errors in beta 3.0.21