Lucifer06/RemoteGPIO #16: issues with v3.0.23 and latest SetupHelper beta

kwindrem, 2024-03-10:
RemoteGPIO is still installing MbItemDigitalInput.qml which makes the install fail because GuiMods is also installing that file. 
( Because packageDependencies also specify that GuiMods must be installed in order for RemoteGPIO to be allowed, the install is blocked whether or not GuiMods is installed. conflicts now displayed in the Package editor menu alternate between GuiMods needing to be installed and uninstalled. )

Note: MbItemDigitalInput.qml in GuiMods now includes the Generic I/O selection.

fileList, fileListVersionIndependent and packageDependencies do not have an end of line on the last file. I think things are working without this but it makes commands like cat display badly on the console.

Lucifer06, 2024-03-11:
Argh, my mistake, while MbItemDigitalInput.qml was correctly removed from the FileSets, I forgot to remove its entry in fileList.
That should be now fixed in v3.0.24

Lucifer06, 2024-03-11:
fixed

closed 2024-03-11 by Lucifer06