Lucifer06/RemoteGPIO #19: SetupHelper v6.13 release approaching

kwindrem, 2024-03-14:
I am getting ready to release SetupHelper v6.13 to the wild.

I just pushed v6.13~8 to the beta branch if you want to look at the latest changes.

Anything else I need to address first ?

(Also posted this issue in SetupHelper repo but wanted to make sure you were aware of it.)

Lucifer06, 2024-03-14:
I have tested v6.13~8 on a fresh installation.

First was install from ssh of v6.10 then it asked for GUI restart.
Then from GUI I switched to beta branch and installed v6.13~8.
It didn't asked for a GUI restart, it went straight away with the GUI restart.

Then I requested the install of RemoteGPIO. Conflicts showed that GuiMods was missing, and not in the active list.
I just added GuiMods (no install) and went back to the conflict resolution.
I thought it would have downloaded and installed it automatically, but it  didn't happen.
[
[image]
](url)

From there I was in kind of dead loop: 
If I go to GuiMods page I get the following page (same if I go to RemoteGPIO): it shows SetupHelper page, not the correct page:
[image]

A restart of PackageManager (from the menu) solved this temporarily issue and I could download & install GuiMods, and then finish RemoteGPIO install.

Except that, it worked like a charm from my perspective.

kwindrem, 2024-03-14:
SetupHelper install does not defer GUI restarts since PackageManager restarts on package install and the state to prompt for the reboot. Not sure I can fix this but will look at it.

Conflict resolution does not download packages. I actually had not considered that but it is possible.
In any case, the lockup is related to a pending install of a package that is not in /data. Great catch, thanks!!!

Is downloading automatically a good thing or not? I will need to address the situation that the internet is not available !

Thanks for testing.

Lucifer06, 2024-03-14:
I don't think the automatic download is required. the conflict message is clear, and user should be able to do the right action for resolving the issue.
But the automatic add of the right package is a good thing, especially when/if we reach a point where there dozens of packages listed in the repository.

kwindrem, 2024-03-14:
Proceeding with conflict resolution that included installing a package that was not on the system locked up the process so I needed to address that.

In the end I got download + install working for conflict resolution. The prompt indicates what will happen before Proceed is pressed so I think it is OK that way.

If the conflicting package is not installed and the GitHub version isn't available for download, I note that to and hide the Proceed button.

kwindrem, 2024-03-15:
It's released not as v7.0.

closed 2024-03-15 by kwindrem