Lucifer06/RemoteGPIO #30: Digital Input doesn't work with RaspBerry

Lucifer06, 2024-04-06:
Relays confirm to work with RaspBerry, but Digital Input won't. It will require additional work for making Digital Inputs working with RaspBerry.
This was tested with Venus 3.40-2 and RemoteGPIO 3.2.2

cbelcher, 2024-04-16:
You're free France email address has been sending me NDR's for days now.  Hit me with a real email.  I'm so close to having this working.  Have one question.

Been searching what creates the directory, for example: /data/RemoteGPIO/sys/devices/platform/soc/1c20800.pinctrl/gpiochip0/gpio/gpio205

Answer me that and I'm sure it'll work :)  

Hope you enjoyed Vegas!  Cheers

Lucifer06, 2024-04-16:
Hello Craig, don't see any issue from my side with my free.fr email. Anyway here is an alternate one if needed: ***@***.*** The /data/RemoteGPIO/sys/devices/platform/soc/1c20800.pinctrl/gpiochip0/gpio/gpio205 is coming from the package itself.  I created these folders manually and are downloaded as a zip with the package (see in the install script). Then I  creates links from these to the /dev/GPIo, depending on GUI configuration. These links are created with the /etc/rcS.d/S90rgpio scriptHope it helped,FreD. Excuse possible typo: swallowed by my iPhone

cbelcher, 2024-04-16:
Oh there's something wrong :)  GitHub Stripped your address and mine, thank goodness.  Send me one from your alternate address.

cbelcher, 2024-04-17:
Now that I have this thing working on the Pi, noticed that dbus_digitalinputs.py, is throwing an exception.  Not often, 
5 times in 1 hour and 10 minutes of uptime. 

Traceback output:

@40000000661fd8230c4843a4   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 614, in poll
@40000000661fd8230c48572c     for inp, level in pulses():
@40000000661fd8230c4862e4   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 169, in __call__
@40000000661fd8230c486e9c     v = int(fp.read())
@40000000661fd8230c496c84 ValueError: invalid literal for int() with base 10: b''
@40000000661fd823120de12c *** starting dbus-digitalinputs ***

trace give you any clues what's going on?
 
Looks like it has something to do with the DT-R008 and its Input registration status?  Been trying to catch it with dbus-monitor, but I don't see anything that is not an integer being returned.  I must have my filter set incorrectly to catch it.

Other than that, it works perfectly.  I'll find it eventually, if your busy.

Having fun learning all this new stuff.

cbelcher, 2024-04-25:
> If you look in Kevin's package manager helper resources he has code in there for checking what platform is being used. His code covers all platforms.

If found where he looks at /etc/venus/machine for the platform in CommonResources, but he’s just looking to see if it’s a RPi or not.  I’ll keep hunting for an actual list.  But sure is a better way to check the platform, then my elaborate d-bus call.

cbelcher, 2024-04-25:
I found this some time ago.  Bug…?  Not consistent, for sure.  

It’s in the S90rgpio_pins.sh script.  

In the Create conf files , Handle Module 1 section.

You deal with Relay’s correctly, since in starts at 3.  But DI’s start at 5, on Cerbo's.  You handle DI - 10 correctly and set to “a”.  DI 11 & 12 not so much.

Output of the Digital_Inputs_unit1.conf.

Goes: 9, a, 11, 12.

/dev/gpio/digital_input_5/value
/dev/gpio/digital_input_6/value
/dev/gpio/digital_input_7/value
/dev/gpio/digital_input_8/value
/dev/gpio/digital_input_9/value
/dev/gpio/digital_input_a/value
/dev/gpio/digital_input_11/value <-  b?  I’ve updated my script, and it works fine as a b.
/dev/gpio/digital_input_12/value < - c   Same deal here.  

I’ve got to figure out how to get my S90 on GitHub.  It’s code or figure out this dang Microsoft GitHub, not a big MS fanboy.  

Works like a champ on RPI 3, 4B and Cerbo.

But all my work is in vein, if I can't get it to you guys.

drtinaz, 2024-04-25:
Good catch. I'll get that fixed straight away.

drtinaz, 2024-04-25:
@cbelcher I have made the changes to S90 and issued a PR. @Lucifer06 can merge and create a new release.

cbelcher, 2024-04-26:
Nice!

Lucifer06, 2024-04-26:
Well-done guys!
PR applied on v3.3.2

cbelcher, 2024-04-26:
> @cbelcher<https://github.com/cbelcher> can you confirm that on the shelf version 3.3.2 is working on your Raspberry (both relays and Digital Inputs), with no further modification?

You bet; I’m coding as I type this.  Keep you posted.

cbelcher, 2024-04-26:
No it does not.  RemoteGPIO v3.3.2 has killed my RPi fix.  Been trying to tack it down.  I see the changes to S90 and moving it to the rcS.d directory.  I've dealt with them, just nothing not even my Relays work any more.  Firing up the Cerbo and see what's different.  
Where do I look for all the modified files from 3.2.2    I'm sure I'll find them if I look hard enough.  With that I can narrow down my target.

Lucifer06, 2024-04-26:
The changes was on S90Let me know what the changes you are making for having the rpi working and I will add it to the standard repo so you will enjoy any new updates with no pain...Excuse possible typo: swallowed by my iPhone

cbelcher, 2024-04-26:
Maybe I missed something, I’m on it.

cbelcher, 2024-04-27:
It's the same thing I've been fighting all day, everything seems to work from the Remote Console configuration standpoint.  Define the Digital input, it shows up in the Device List.  But when I short the input the status does not register.

I watch the dbus-monitor and no longer do I see the messages like I use too.  Like this is from the last working version 3.2.1. 

Neither 3.2.2 or 3.3.2, work.  I see nothing on the bus. 

`signal time=1713282122.199821 sender=:1.49410 -> destination=(null destination) serial=26186 path=/; interface=com.victronenergy.BusItem; member=ItemsChanged
   array [
      dict entry(
         string "/InputState"
         array [
            dict entry(
               string "Value"
               variant                   int32 0
            )
            dict entry(
               string "Text"
               variant                   string "0"
            )
         ]
      )
      dict entry(
         string "/State"
         array [
            dict entry(
               string "Value"
               variant                   int32 6
            )
            dict entry(
               string "Text"
               variant                   string "open"
            )
         ]
      )
   ]`

The only clue I see is the working v3.2.1 I, have a units.conf in /data/RemoteGPIO/conf.  3.2.2 and 3.3.2, it's missing, only the relays.conf file is in this directory.  What am I missing?  

I see S90 no longer references it, so has me thinking there is something else?

Thoughts?  Need to get a pcap going and see if I see the messages, maybe the RPi just isn't listen to them.

cbelcher, 2024-04-27:
My statement that 3.2.2 doesn't work, is simple false.  I've got notes and console output that proves that it did.  Not much point in dealing with the past.  Have to figure out what's wrong with the current release.

Any ideas, toss them out!

drtinaz, 2024-04-27:
Have you enabled digital inputs in the 'additional options' menu for each relay module?

cbelcher, 2024-04-27:
No Sir, but I have tried it in the past.  Need to do anything after I enable it?  Reboot, etc?

cbelcher, 2024-04-27:
Working..  What the heck is that option there for, lol....  I tried it earlier this morning and it didn't change anything?  Been rebuilt 5 times since then, so who knows.

Okay, we have a fully function RPi 4 Model B running RemoteGPIO v3.3.2.  Nice..

Now I just copy my S90 to my git folder then do what, a commit?

drtinaz, 2024-04-27:
Yes. Any changes to configuration will require a manual restart of RemoteGPIO from within the menu.

cbelcher, 2024-04-27:
You are correct.

drtinaz, 2024-04-27:
> Working..  What the heck is that option there for, lol....  I tried it earlier this morning and it didn't change anything?  Been rebuilt 5 times since then, so who knows.
>
> Okay, we have a fully [...]

The option is there because not everyone will use the feature, and disabling it saves on system resources. Same reason with reading relays option.

cbelcher, 2024-04-27:
Makes sense.  With the PoS Cerbo you need every clock cycle you can get.  I’m still in shock my 5 year old,  $45 RPi has just over twice the MIPS score.

I think I’ll hang it on the wall for the final install and hide the RPi behind the battery racks 😊

Lucifer06, 2024-04-27:
@cbelcher then, did you had to tweak S90 script (or anything else) or the latest v3.3.2 is working fine on a stock rPI?
Cheers,
FreD.

cbelcher, 2024-04-27:
Only changes to S90 are needed to get the rPI working.

Going to spend the morning, trying to get the code to my GitHub fork.

cbelcher, 2024-04-27:
Just ran through a complete, from a blank SD card to a fully functional RPi 4 Model.  Running RemoteGPIO v3.3.2, with a single DT-R008.
Firmware & Packages
Firmware: 3.40~10
SetupHelper: v7.18
GuiMods: v10.24
ShutdownMonitor: v3.2
RpiTemperature: v1.2
RemoteGPIO: v3.3.2

Just have to pause, after installing RemoteGPIO when it's asking you to reboot.  Shim in my S90, then select now.
Configure RemoteGPIO.  Bam!  Just works.

Let me get this very old version of my S90 script committed on GitHub, long story why I chose an old version.  I'll work on getting the much slicker version published this evening.

The key to getting it working, is that the stock S90 assumes that dbus_digitalinput service is running.  Well on any device without native Digital Inputs and that list includes several Victron devices.  start-digitalinputs.sh puts a stop to that.  It does a test for the existence of /dev/gpio/digital_input_1.  Doesn't find it, it takes the service down.  For any device without native DI's I bring it up.  You'll see when I can figure out how to do a commit, YouTube to the rescue.

Most recent version of the script handles all the devices without native DI's and is a 100 or so lines smaller.  This version was just the first that worked, it isn't pretty.  Other biggie, this one I only had coded for Module 1.  Once I got it working I was off to optimizing it, then I dealt with the other modules.

cbelcher, 2024-04-28:
@Lucifer06 @drtinaz 

Did you see my commit?  Hope I didn't mess anything up.

cbelcher, 2024-04-28:
Looks like you guys got it, I see it on both your GitHub pages.  I'll clean 90% of it up tomorrow, and we can have the the all the RPi users back online.

cbelcher, 2024-04-28:
Just tested on Cerbo and there is an issue.  I'll find it quick.

cbelcher, 2024-04-28:
I found it, it's my platform detection call.  Need change my get_platform_setting sub to awk for $1 not 3.

Please hold.

drtinaz, 2024-04-28:
Are you starting with the current S90 script file and making your changes to that? Using a file from an older version may lose some functionality or break something.

cbelcher, 2024-04-28:
Fact is I don’t have a dam clue how to use GitHub.  I’m trying,  but I knew there would be something I would do incorrectly.

Got the relay’s working on my Cerbo GX MK2, but DI’s are not happy.  Service dbus-digitalinputs is crashing every 3 seconds.  Even after uninstalling and reinstalling.  It obviously leaves a trail of changes after uninstalling.

The second I uninstall it, crashes stop, install it they start.  Using 3.3.2 S90, does the same thing.

cbelcher, 2024-04-28:
I’ve just completed a full reimage of my Cerbo GX MK2, and dbus-digitalinputs service is tracing back just like with my S90.

You sure this is rock solid?  It behaves the same way as my modified S90?

Tracebacks every 3-4 seconds.
Whatever is going on it’s not my S90…

***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15503) 2 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15591) 1 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15591) 2 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15591) 3 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15591) 4 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15661) 1 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15661) 1 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15661) 2 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15661) 3 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15661) 3 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15751) 0 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15751) 1 seconds
***@***.***:~#  svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 15751) 2 seconds

@40000000662ea05829874b54 *** starting dbus-digitalinputs ***
@40000000662ea05a16c2a70c Registering GPIO 1 for type 0
@40000000662ea05a230a9e84 Registering GPIO 2 for type 0
@40000000662ea05a2f6239e4 Registering GPIO 3 for type 0
@40000000662ea05b000cbfac Registering GPIO 4 for type 0
@40000000662ea05b0fe9ad54 Registering GPIO 5 for type 0
@40000000662ea05b1ce18824 Registering GPIO 6 for type 0
@40000000662ea05b2988399c Registering GPIO 7 for type 0
@40000000662ea05b37eb161c Registering GPIO 8 for type 0
@40000000662ea05c0a4f752c Registering GPIO 9 for type 0
@40000000662ea05c17d6786c Registering GPIO 10 for type 2
@40000000662ea05c18ec071c Traceback (most recent call last):
@40000000662ea05c18ee4d24   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 651, in <module>
@40000000662ea05c191bf0bc     main()
@40000000662ea05c191e59ec   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 607, in main
@40000000662ea05c1a3501dc     register_gpio(pth, inp, bus, sd)
@40000000662ea05c1a3776c4   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 547, in register_gpio
@40000000662ea05c1a61f5ac     handler.level = pulses.register(path, gpio)
@40000000662ea05c1a64476c   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 111, in register
@40000000662ea05c1a7d78a4     self.ob.register(fp, EPOLLPRI)
@40000000662ea05c1a7ff174 PermissionError: [Errno 1] Operation not permitted
@40000000662ea05c2446c034 *** starting dbus-digitalinputs ***
@40000000662ea05e02fc3a94 Registering GPIO 1 for type 0
@40000000662ea05e0fd2413c Registering GPIO 2 for type 0

Not code of mine ,is running on the Cerbo GX MK2.

cbelcher, 2024-04-28:
These tracebacks indicate some Permissions Error which I don’t see on the RPi.  I’ve seen that if you don’t start from a fresh install, it behaves very differently.

I’ll try to find it, but something is very wrong.

Factory fresh Cerbo GX MK2,
Firmware 3.40~10
SetupHelper: v7.18
GuiMods: v10.24
ShutdownMontior: v3.2
RemoteGPIO: v3.2.2

Flat out does not work!

Works on my Pi with my shimmed S90..  Sure, I’ve been concentrating on that platform.  Just assumed the Cerbo GX was solid.  Not so much.

cbelcher, 2024-04-29:
Digging deeper on this complete stock v.3.3.2 install on the Cerbo GX MK2.  With Unit 1 configured with 8 relays, digitalinput service traces every 3-4 seconds.  Set unit 1 to 2 Relays and it no longer traces.  Going to try 4 Relays next, betcha it works.  

If you watch dbus-spy this input09 is the culprit.  It is displayed for a ms in "com.victronenergy.system" and then vanishes.

Like so:  I'm scrolling back in dbus-spy to see what was being displayed.
`com.victronenergy.digitalinput.input09                                                                            
com.victronenergy.digitalinput.input09                                                                            
com.victronenergy.digitalinput.input09                                                                            
com.victronenergy.digitalinput.input09                                                                            
com.victronenergy.digitalinput.input09                                                                            
com.victronenergy.digitalinput.input09                                                                            
com.victronenergy.system      

com.victronenergy.adc
com.victronenergy.fronius                                                                                         
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.packageManager                                                                   Package Manager
com.victronenergy.platform                                                                                   Venus
com.victronenergy.settings
com.victronenergy.shutdown                                                                         ShutdownMonitor
com.victronenergy.system      
com.victronenergy.vecan.vecan0
com.victronenergy.vecan.vecan1`

2 Relays the service has not traced a single time
`@40000000662f993f14372db4 *** starting dbus-digitalinputs ***

@40000000662f994019125f84 *** starting dbus-digitalinputs ***
@40000000662f994216bede4c Registering GPIO 1 for type 0
@40000000662f99422a62b34c Registering GPIO 2 for type 0
@40000000662f994302b415d4 Registering GPIO 3 for type 0
@40000000662f994315922dbc Registering GPIO 4 for type 0
@40000000662f994329f7e38c Registering GPIO 5 for type 0
@40000000662f99440294d5d4 Registering GPIO 6 for type 0

root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 13036) **6164 seconds**`

cbelcher, 2024-04-29:
Proved setting Unit 1 to 8 Relays, triggers the tracebacks in dbus_digitalinputs.py.  Set to 2 or 4, no issue.

`root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 25932) 459 seconds

root@cerbogx-mk2:/etc/rcS.d# 
root@cerbogx-mk2:/etc/rcS.d# ############# Above is with Relays = 4, not tracing just about to set back to 8 to prove it's the trigger of the traces ########
 
root@cerbogx-mk2:/etc/rcS.d# date
Mon Apr 29 15:02:38 UTC 2024
root@cerbogx-mk2:/etc/rcS.d# #######changing now to 8 #############
 
root@cerbogx-mk2:/etc/rcS.d# date
Mon Apr 29 15:03:02 UTC 2024
 
root@cerbogx-mk2:/etc/rcS.d# ########Restarting RemoteGPIO Service #################

root@cerbogx-mk2:/etc/rcS.d# date
Mon Apr 29 15:03:40 UTC 2024
root@cerbogx-mk2:/etc/rcS.d# ############Tracing starts immediately ####################
root@cerbogx-mk2:/etc/rcS.d# 
root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 26818) 3 seconds
root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 27054) 1 seconds
root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 27054) 2 seconds
root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 27054) 2 seconds
root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 27054) 3 seconds
root@cerbogx-mk2:/etc/rcS.d# svstat /service/dbus-digitalinputs
/service/dbus-digitalinputs: up (pid 27054) 4 seconds
root@cerbogx-mk2:/etc/rcS.d# 
root@cerbogx-mk2:/etc/rcS.d# ############ Service will not say running ############
root@cerbogx-mk2:/etc/rcS.d# `

Something's up.  I'm on a missing to find it.

cbelcher, 2024-04-29:
> Check the created links in /dev/gpio and the digital input .conf files.

I have they look as they should.
With 8

***@***.******@***.***:/etc/rcS.d#> ls -lah /dev/gpio
drwxr-xr-x    2 root     root         520 Apr 29 15:03 .
drwxr-xr-x   14 root     root        2.9K Apr 28 18:56 ..
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_1 -> /sys/class/gpio/gpio226
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_2 -> /sys/class/gpio/gpio225
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_3 -> /sys/class/gpio/gpio224
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_4 -> /sys/class/gpio/gpio227
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_5 -> /data/RemoteGPIO/sys/class/gpio/gpio205
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_6 -> /data/RemoteGPIO/sys/class/gpio/gpio206
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_7 -> /data/RemoteGPIO/sys/class/gpio/gpio207
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_8 -> /data/RemoteGPIO/sys/class/gpio/gpio208
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_9 -> /data/RemoteGPIO/sys/class/gpio/gpio209
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_a -> /data/RemoteGPIO/sys/class/gpio/gpio210
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_b -> /data/RemoteGPIO/sys/class/gpio/gpio211
lrwxrwxrwx    1 root     root          39 Apr 29 15:03 digital_input_c -> /data/RemoteGPIO/sys/class/gpio/gpio212

***@***.***:/data/RemoteGPIO/FileSets/Conf# cat Digital_Inputs_unit1.conf

/dev/gpio/digital_input_5/value
/dev/gpio/digital_input_6/value
/dev/gpio/digital_input_7/value
/dev/gpio/digital_input_8/value
/dev/gpio/digital_input_9/value
/dev/gpio/digital_input_a/value
/dev/gpio/digital_input_b/value
/dev/gpio/digital_input_c/value

And 4
***@***.***:/data/RemoteGPIO/FileSets/Conf# ls -lah /dev/gpio
drwxr-xr-x    2 root     root         360 Apr 29 15:22 .
drwxr-xr-x   14 root     root        2.9K Apr 28 18:56 ..
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_1 -> /sys/class/gpio/gpio226
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_2 -> /sys/class/gpio/gpio225
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_3 -> /sys/class/gpio/gpio224
lrwxrwxrwx    1 root     root          23 Apr 28 18:55 digital_input_4 -> /sys/class/gpio/gpio227
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 digital_input_5 -> /data/RemoteGPIO/sys/class/gpio/gpio205
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 digital_input_6 -> /data/RemoteGPIO/sys/class/gpio/gpio206
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 digital_input_7 -> /data/RemoteGPIO/sys/class/gpio/gpio207
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 digital_input_8 -> /data/RemoteGPIO/sys/class/gpio/gpio208
lrwxrwxrwx    1 root     root          23 Apr 28 18:56 mkx_rst -> /sys/class/gpio/gpio122
lrwxrwxrwx    1 root     root          21 Apr 28 18:55 relay_1 -> /sys/class/gpio/gpio4
lrwxrwxrwx    1 root     root          21 Apr 28 18:55 relay_2 -> /sys/class/gpio/gpio5
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 relay_3 -> /data/RemoteGPIO/sys/class/gpio/gpio103
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 relay_4 -> /data/RemoteGPIO/sys/class/gpio/gpio104
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 relay_5 -> /data/RemoteGPIO/sys/class/gpio/gpio105
lrwxrwxrwx    1 root     root          39 Apr 29 15:22 relay_6 -> /data/RemoteGPIO/sys/class/gpio/gpio106
lrwxrwxrwx    1 root     root          23 Apr 28 18:56 vebus_standby -> /sys/class/gpio/gpio123
***@***.***:/data/RemoteGPIO/FileSets/Conf# cat Digital_Inputs_unit1.conf

/dev/gpio/digital_input_5/value
/dev/gpio/digital_input_6/value
/dev/gpio/digital_input_7/value
/dev/gpio/digital_input_8/value

Trace hit when it’s registering GPIO 9, which doesn’t come into play with either Relays set to 2 or 4.  8 Yes.
@40000000662fbb350bb27be4 Registering GPIO 9 for type 2
@40000000662fbb350e5f293c Traceback (most recent call last):
@40000000662fbb350e619a3c   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 651, in <module>
@40000000662fbb350edc63d4     main()
@40000000662fbb350edf0f6c   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 607, in main
@40000000662fbb350f2fc77c     register_gpio(pth, inp, bus, sd)
@40000000662fbb350f322cc4   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 547, in register_gpio
@40000000662fbb350f95671c     handler.level = pulses.register(path, gpio)
@40000000662fbb350f97fb44   File "/opt/victronenergy/dbus-digitalinputs/dbus_digitalinputs.py", line 111, in register
@40000000662fbb350fad82fc     self.ob.register(fp, EPOLLPRI)
@40000000662fbb350fafb57c PermissionError: [Errno 1] Operation not permitted

cbelcher, 2024-04-29:
Is see what’s causing it?  I know that startdigitalinputs.sh looks for the /edge file.  It exists in gpio8 but not 9.
210 it’s back, 211 missing, 206 & 207 not there.  It’s all over the map.

I’ve experimented with this long ago.  Must looks though my notes, it was relevant for sure.

Keep you posted.

cbelcher, 2024-04-29:
I see these edge files were created, when I set the Digital Inputs 9 and 10 from Disabled to Door alarm.  Time stamps match up.

`root@cerbogx-mk2:/data/RemoteGPIO/sys/class/gpio/gpio209# ls -lah

drwxr-xr-x     3 root     root        4.0K Apr 28 19:19 .
drwxr-xr-x   34 root     root        4.0K Feb 23 16:23 ..
-rw-r--r--      1 root     root            2 Feb 22 23:14 active_low
lrwxrwxrwx    1 root     root          18 Feb 22 23:14 device -> ../../../gpiochip0
-rw-r--r--       1 root    root            3 Feb 22 23:14 direction
-rw-r--r--       1 root    root      70.4K Apr 29 15:46 edge             < ---  When I Enabled DI 9, and all heck broke loose.`

Mmmm..... Let's see if I can figure out what is going on.

cbelcher, 2024-04-29:
The edge file on gpio209 is 70K, pages of "bothbothboth....."  210 is only 1.1K with the same thing bothboth, at least 100+ times.
Not normal.

cbelcher, 2024-04-29:
I see every time it traceback's, it appends a "both" to the edge file.  Very odd.

It's got to be the "dbus_digitalinputs.py" script with the date stamp when I installed RemoteGPIO?

cbelcher, 2024-04-29:
Yup, sure does.  It's a symptom though, not the cause of it tracing back.  What the heck is going on.

cbelcher, 2024-04-29:
So I've proved what causes /service/dbus-digitalinput to traceback on my bone stock Cerbo GX MK2.
Firmware: v3.40~10
SetupHelper: v7.18
GuiMods: v10.24
ShutdownMonitor: v3.2
RemoteGPIO: v3.3.2

 If you configure any Digital input that exists on the DT-R008, it traces immediately.  If you disable them the service run forever.  Configure native DI's 1-4, no issue.  

Going to rebuild this from scratch and see if I have the same results, if so it maybe it's my Cerbo MK2?  We shall see.
 
 Can one of you list your Firmware and Package versions your running, that you know works?

Lucifer06, 2024-04-29:
On my production CerboGX MK1, I'm running 3.31 with latest RemoteGPIO package. On my Dev Cerbo GX MK1, I'm running latest venus beta (3.4.x) and latest RemoteGPIO.
Both working perfectly

cbelcher, 2024-04-29:
Total OS rebuild on Cerbo GX MK2.  Same thing, if this works for you guys it's got to me my MK2?
Relays work, local Digital Inputs work, but remote DI's DO NOT work on my Cerbo.  Set DI 5 to Door alarm and /service/dbus-digitalinputs traces instantaneously.

I have everything documented, I'll post the build and show you it tracing.  Have to go have dinner with mom.

I thought I was nuts, but 2 times same results.

cbelcher, 2024-04-29:
Build notes from 2nd attempt.

[Total rebuild of Cerbo 2nd time.txt](https://github.com/Lucifer06/RemoteGPIO/files/15155566/Total.rebuild.of.Cerbo.2nd.time.txt)

Anything look odd?

cbelcher, 2024-04-29:
Yes, this run I kept Official Large 3.31, latest RemoteGPIO v3.3.2.  Last run was with Beta 3.40~10 large, no change.

I’ve seen both the Cerbo and Rpi’s behave very differently when you start from ground zero.  Can you also start with a factory default Cerbo and see if you can get it running.  It’s the only way I see to test my theory.  If it works, then it’s the MK2, if it doesn't (which I don’t believe it will) the SoC is identical to the to the MK1, they’ve flipped some ports orientation’s and changed a single USB-Type A to support data.  It’s possible it’s the MK2, but I think unlikely.

The /data/logs/dbus-digitalinputs/current.  Don’t look right.

See my “Total rebuild of Cerbo 2nd time.txt”
Lines 658 – end

Line 707 shows 4 Registering, all 4 native GPIO’s 1-4
712 is where I change GPIO 5 to Door alarm type 2.
I would have had to have configured RemoteGPIO to 1 Unit and 8 Relays at this point.  In turn, would have expective the logs to show all 12 Registered before this point, 4 natives + my 8 DT-R008’s.  Nope only 4.

Line 726 you see the traceback with a PermissionError, this is where I set DI 5 to Door alarm.
Then back to registering 1-4, hits 5 and traces, every 3-4 seconds until I clear it.

930 where I set DI 5 to Disable. only then do I see all 12 Register.

There is an answer to this, just need to find it.  And proving it’s not my MK2 would be super helpful.

Let me know, I’ll buy a MK1 if I have to.

cbelcher, 2024-04-30:
Tired mapping around 205-208, still traces.  I won’t stop until I find it 😊

cbelcher, 2024-04-30:
No matter where I map the gpio2xx to on the Cerbo. it creates this edge file.  The RPi does not.

It cannot be there.  Will never work, as it traces it just appends a “both” to the file until I stop it.

Cerbo GX MK2 after mapping digital_intput_5 to 209, did the same thing on every mapping.  Including the default gpio205.
***@***.***:/dev/gpio/digital_input_5# ls -lah
drwxr-xr-x    3 root     root        4.0K Apr 30 00:18 .
drwxr-xr-x   34 root     root        4.0K Feb 23 16:23 ..
-rw-r--r--    1 root     root           2 Feb 22 23:14 active_low
lrwxrwxrwx    1 root     root          18 Feb 22 23:14 device -> ../../../gpiochip0
-rw-r--r--    1 root     root           3 Feb 22 23:14 direction
-rw-r--r--    1 root     root          20 Apr 30 00:18 edge  <-- 1 minutes it’s already 20 bytes, shouldn’t even be there.
lrwxrwxrwx    1 root     root          34 Feb 23 15:18 gpio209 -> /data/rgpio/sys/class/gpio/gpio209
drwxr-xr-x    2 root     root        4.0K Feb 22 23:14 power
lrwxrwxrwx    1 root     root          31 Feb 22 23:14 subsystem -> ../../../../../../../class/gpio
-rw-r--r--    1 root     root           0 Feb 22 23:14 uevent
-rw-r--r--    1 root     root           2 Apr 30 01:02 value
***@***.***:/dev/gpio/digital_input_5# cat edge
bothbothbothbothboth
***@***.***:/dev/gpio/digital_input_5#

RPi
***@***.***:/dev/gpio/digital_input_5# ls -lah
drwxr-xr-x    3 root     root        4.0K Feb 23 15:18 .
drwxr-xr-x   34 root     root        4.0K Feb 23 16:23 ..
-rw-r--r--    1 root     root           2 Feb 22 23:14 active_low
lrwxrwxrwx    1 root     root          18 Feb 22 23:14 device -> ../../../gpiochip0
-rw-r--r--    1 root     root           3 Feb 22 23:14 direction
lrwxrwxrwx    1 root     root          34 Feb 23 15:18 gpio209 -> /data/rgpio/sys/class/gpio/gpio209
drwxr-xr-x    2 root     root        4.0K Feb 22 23:14 power
lrwxrwxrwx    1 root     root          31 Feb 22 23:14 subsystem -> ../../../../../../../class/gpio
-rw-r--r--    1 root     root           0 Feb 22 23:14 uevent
-rw-r--r--    1 root     root           2 Apr 30 01:24 value
***@***.***:/dev/gpio/digital_input_5#

No, edge file.  Works!

I’m on to something, just need to divide and concur to rule out what it’s not.

cbelcher, 2024-04-30:
@Lucifer06 @drtinaz 

I know I've asked you before if you were familiar with the /opt/victronenergy/dbus-digitalinputs/dbus-digitalinputs.py script where you added the 'Transfer switch'.  This is the script creating/appending the "both" to the edge file every time dbus-digitainputs service starts.  The edge file is not created on the RPi and works.  

The file is created here.   Still can't figure out how to get this thing to format like I want it to :(

`class EpollPulseCounter(BasePulseCounter):
    def __init__(self):
        self.gpiomap = {}
        self.states = {}
        self.ob = epoll()

    def register(self, path, gpio):
        path = os.path.realpath(path)

        # Set up gpio for rising edge interrupts
        with open(os.path.join(path, 'edge'), 'ab') as fp:
            fp.write(b'both')`

Guess I need to query dbus to see what's different between the 2 platforms.  Cerbo the edge file is created and the RPi it's not.  

Still don't know if this is my MK2 or not, but the fact is RemoteGPIO Digital inputs DO NOT function on it.  Relays no issue.

Curious, either of you guys have edge files in gpio205 if you enable Input 5 on your MK1's?

I may need some help on this one?

cbelcher, 2024-04-30:
@Lucifer06 @drtinaz

I see what's going on.  Can you either of you check your Cerbo's to see if the flags parameter is set in the dbus_digitalinputs.py process?  It should not.  It will show up between "dbus_digitalinputs.py **(here)** /dev/gpio/digital_input_1"

Like this.

RPi
`dbus_digitalinputs.py --poll=poll /dev/gpio/digital_input_1`

Cerbo GX MK2
`dbus_digitalinputs.py /dev/gpio/digital_input_1`

If that flag is not set to "--poll=poll" there is no way the remote DI's can function.  

Going to prove it right after this.  It's all in the start-digitalinput.sh file.  Haven't tested, but I'll the farm that the Native DI's will work on my Cerbo.  Since they require the edge file to be present and set to "none".

And while you're at it, tell me if gpio205 has the edge file present.

cbelcher, 2024-04-30:
> Guimods is responsible for modifying the digital inputs python script so you might hit up @kwindrem<https://github.com/kwindrem> and see if he has any insight.

I think I’ve got it, not a solution but to prove my theory.  If it holds, then the problem definition is at hand.

Keep you posted, will know in about 5 minutes.

D, you in front of your Cerbo?  Or you at work?

drtinaz, 2024-04-30:
I have just finished migrating all of my automation's over to a Rpi4 running home assistant OS with node red integrated. Home assistant is now in full control of everything, though it took a few days to accomplish. I was forced into a crash course of MQTT communications between the cerbo and HA. I am no longer running RemoteGPIO natively on the GX and I don't currently have my test system setup. I don't have an 8 relay module so I would't be much help in that regard anyway. I am now using TCP/IP through node red which integrates the relay settings and overview in the Cerbo via MQTT.

On a side note, I am extremely happy with my setup now. Home assistant is by far a much nicer GUI and the cerbo is now restricted to only doing cerbo things which has resulted in a much snappier and more responsive experience. I have a lot of Victron components connected to the cerbo and it really was not designed to handle a lot of extra load put on it by all these add-ons.

kwindrem, 2024-04-30:
My RPI4 has the ...edge... file but my Cerbo does not. I'm sure it's a MK1 Cerbo.

I doubt that the Dingtian module would provide pulse-counter functionality. If not, you may need to modify other parts of the digital input code to insure it's not selectable and to insure the functions associated with it don't get called.

cbelcher, 2024-04-30:
@Lucifer06
Theory proved,

I've got both native and remote Digital inputs working on Cerbo GX MK2.  No more tracebacks every 4 seconds, no more edge files created on the remote gpio devices.

I have the solution, but you guys have to let me know if you can reproduce this on your dev Cerbo MK1's.  That will entail a complete Venus OS reinstall and build it back from scratch.  I can't see how it's going to work?  Take 30 minutes max.

Duplicate mine:
Firmware: v3.31 Official release
Type: Large
Node-RED Enabled
SetupHelper: v7.18
GuiMods: v10.24
ShutdownMonitor: v3.2
RemoteGPIO: v3.3.2
1 - Dingtian DT-R008
Hardware Version: V3.6J
Software Version: V3.1.3541B

Shortly, there are going to be MK2's all over the place.  We have to figure out if this is isolated to just the MK2.

cbelcher, 2024-04-30:
@kwindrem, "The Master" joins the thread.  Thanks Kevin!

I'd been working on getting these Digital inputs working on the RPi, then I noticed that my code didn't run on my Cerbo.  I was like what the heck, didn't change a single line of code that dealt with the Cerbo GX.  Did a complete OS reload, and still didn't work.  Just 1/2 hour ago I got it working.  It's the startdigitalinputs.sh script not setting the flags=--poll=poll then calling dbus_digitalinputs.py.  If that flags parameter is null, it causes all sorts of issues with remote DI's.  One of which is, they don't work :)

Boy, been fun learning how this all works.

I'm a HomeSeer guy, have been for 15+ years.  Tried out HA, just didn't do anything for me.  So new to all this, that integration is way down on my list.

Thanks for chiming is!
Cheers.

cbelcher, 2024-04-30:
The comments in dbus_digitalinputs.py are pretty funny.  

`            # NOTE: There has not been a report of a missed interrupt yet.
            # Belts and suspenders.`

Least he's having fun...

@kwindrem
What's --poll=poll, parameters role in this script?  It's getting pretty deep, can you sum it up in a sentence or 2?

kwindrem, 2024-04-30:
I am by no means the master of digital inputs code. My best guess is that poll has to do with how information about some part of a digital input is obtained. Based on the comment you quoted, my best guess is that some systems use interrupts and some have to pole for changes.

I don't know what will happen if you enable poll on systems that aren't currently using it. Might cause issues with the internal digital inputs. So you might need to put polling for the remote digital inputs somewhere else or add code to see if it's a local or remote digital input and base polling on that if poll is not set.

I know the Victron engineers are busy with gui-v2 but might be prudent to ask for some understanding.

Sorry I can't be of any real help here.

cbelcher, 2024-04-30:
@kwindrem

With it set to --poll=poll, native digital inputs show no signs of issues.  The di.py script, doesn't even list it as an option.  

`parser.add_argument('--poll',
        help='Use a different kind of polling. Options are epoll, dumb and debug',
        default='epoll')`

But for sure if I don't send it the $flags parameter and epoll is the default.  It flat out will not detect the state change of a remote digital input.  

Thanks again Kevin!  Putting this to bed.

Cheers!

gharshaw, 2024-10-15:
Looking to add this to my RPI but need digital inputs for generator control. Are digital inputs still not working with the RPI's?

[13 more replies]

closed 2025-07-23 by Lucifer06