Lucifer06/RemoteGPIO #38: Error installing RemoteGPIO on new Cerbo

StefanVerbiest, 2024-05-27:
Hi,

Hope I am doing this like it should, I am a newby at Git..

I had a bricked Cerbo GX so I ordered a new one first and I tried to install all my packages on the new Cerbo:

Package Manager
GuiMods.

This all went well.

When I try to install RemoteGPIO on the new Cerbo I got this:

@400000006654749e12d2cf3c WARNING:adding RemoteGPIO
@40000000665474b40a109f3c WARNING:downloading RemoteGPIO
@40000000665474b52b60aaec WARNING:checking RemoteGPIO
@40000000665474b60005ddf4 WARNING:RemoteGPIO check failed - unknown error 1

After that I flashed my old Cerbo and tried to install all again, same output.
This is on FW 3.30 and the latest RemoteGPIO.

Now I have flashed my Cerbo with FW 2.94 and installed all packages and RemoteGPIO v2.4.2 and that works.
After updating the Cerbo to FW 3.32 I try to update REmoteGPIO, same error.

I hope you have enough info for this.

KR Stefan

StefanVerbiest, 2024-05-27:
I looked a bit further and found this:

RemoteGPIO: helper file LogHandler not found - can't continue
WARNING: this log file no longer used
      SetupHelper now logged to /var/log/PackageManager/current
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
WARNING: this log file no longer used
      SetupHelper now logged to /var/log/PackageManager/current
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
WARNING: this log file no longer used
      SetupHelper now logged to /var/log/PackageManager/current
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
WARNING: this log file no longer used
      SetupHelper now logged to /var/log/PackageManager/current
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
RemoteGPIO: helper file LogHandler not found - can't continue
GuiMods: helper file LogHandler not found - can't continue
WARNING: this log file no longer used

drtinaz, 2024-05-27:
Setuphelper has some major changes between v7.xx and V8.xx. The helper files code in the setup script of RemoteGPIO will need to be updated and a new version released. I'm traveling the next several days so I don't have time to do it right now. Hopefully Frederic can get to it. In the meantime you can roll back setuphelper to v7.10 and then install RemoteGPIO again.

@Lucifer06 if you have a chance to get to it before I do, look at the generic setup script provided in the latest setuphelper package for the new method of calling the helper files that is what will need changed.

StefanVerbiest, 2024-05-27:
Thank you very much for this quick answer and solution. It did the trick over here.

Kr Stefan

Lucifer06, 2024-05-27:
I'm travelling in middle east for 2 weeks.Hope I can work on it mid week...Cheers,FreDExcuse possible typo: swallowed by my iPhone

Lucifer06, 2024-05-28:
> @Lucifer06 <https://github.com/Lucifer06> Thank you for the quick fix in your package. It works now, so great :D
> One little thingy, your version is now v8.3=4 ;)
>
> Have a great trip in the Middle East and thanks again.

I haven’t yet done anything… ;-)

------------------------------------------------------------------------------
The water is always bluer on the other side of the ocean ;-)

drtinaz, 2024-05-28:
@Lucifer06 Kevin released a fix in v8.4 that's supposed to fix the backwards compatibility problem, so you might not need to do anything.

closed 2024-08-24 by Lucifer06