Lucifer06/RemoteGPIO #43: GuiMods Dependency Reasons?

Seneda, 2024-09-11:
Is it possible to have an option to run this plugin without GuiMods? (For my use case i do not want to have the modified screens etc. but wish to control the relays separately via a custom plugin)
I plan to make a forked version, but thought I would check why the dependency on GuiMods exists?

Lucifer06, 2024-10-01:
The dependencies exists, not only for the Relay page, but also to populate the dbus and have the menus to display the additional  relays and DI

tempesta-home, 2025-01-05:
Hi Lucifer06, first of all thank you very much for your sw.  
I was wondering if you can publish a "only backend" version without the guimods dependency.  
If you don't have to worry about gui maybe it could be simpler
Someone could use the node-red dashboard to drive the switches and to display the digital inputs status...
What do you think about it?
Thx

Lucifer06, 2025-01-09:
Hi, the request for backend makes senses, and actually, I'm more or less after it for making it working with V2 Victron GUI.
I'll post this when I'm ready, but be patient as I have many plates on my table....

Lucifer06, 2025-08-01:
Closing the issue with the release of 4.0.0