Lucifer06/RemoteGPIO #45: Significant delay on relays using Cerbo GX with the USB->RS485 connection

onnimonni, 2024-09-30:
Hey,

The native relays on Cerbo GX are very snappy and activate instantly.

Also when I'm using the Dingtian IOT Relay web admin the external relays activate instantly.

But when I'm using the Dingtian through Cerbo GX there's very significant and random delay from 4s up to 25s.

What could I do to decrease the delays on the external board?

I'm using the latest RemoteGPIO 3.3.3 and Venus OS 3.4.2 and the recommended relay board and the RS485 adapter from Aliexpress.

I saw that you had similiar issues on #22 and the root cause was fixed on earlier versions.

Lucifer06, 2024-10-01:
Hi, 
The delay with Dingtian relay box will not react as fast as the internal 2 relays of the Cerbo. 
The relay is coming from the driver itself, the internal dbus latency to show up updates, the modbus communication protocole, the limited cpu resources available...
That said, the 4s to 25s dely seems excessive!
Have you tried to reboot the Cerbo?
Also depending on how many devices are attached to the Cerbo, you may see latency increasing latency if you add many MPPT regulators, BMS, etc...
Also try to disable the unused services. For ex Node-red consumes à lot of CPU. 
The latency should not exceed 1s

onnimonni, 2024-10-01:
I don't have anything else plugged to the Cerbo other than the victron touch screen and the dingtian IOT relay.

onnimonni, 2024-10-01:
The delay is very random for me. Sometimes it's quite snappy but sometimes it takes for a very long time. You can see it in the attached video:
https://github.com/user-attachments/assets/c61eb051-e194-4a3b-8324-fe6270412044

I'm happy to provide more logs if you can just point where I should copy them 👍

onnimonni, 2024-10-01:
>Configure the Dingtian to also work in Ethernet. There are config to be done on its web gui, and also on RemoteGPIO config page.
If no delay then you know this is related to USB0.

Can I plug it directly to the Cerbo GX ethernet port or do I need to have a switch/router in between them?

>I strongly recommend to go with a RS485 to USB adapter based on FTDI chipset, in particular FT232R and follow the instructions on how to program it on the Victron Community RemoteGPIO page.
Unfortunately Victron has not migrated the articles from the old community, but you find it here:
https://communityarchive.victronenergy.com/questions/245584/remotegpio-ex-venus-rgpio-now-available-with-setup.html

I did buy the recommended RS485 -> USB adapter from Aliexpress but I did not yet modify it in anyway. This could be the reason.

Lucifer06, 2024-10-01:
Yopu need to program the FTDI chipset. That's the reason because then the default /etc/udev/rules.d/serial-starter.rules will not find the RS485 adapter to be assigned to RemoteGPIO.
That's I think the reason.

onnimonni, 2024-10-01:
Thanks!

Hmmm I think the [USB->RS485 adapter I bought](https://www.aliexpress.com/item/1005004520479272.html) doesn't have FTDI chipset. I'll order a new one. Can you point me to one that you would recommend?

Lucifer06, 2024-10-01:
the one you have is CH41 based.
You can make it working by making sure it is connected to 1st USB port, and uncomment the line
ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_PATH_TAG}=="platform-1c14400_usb-usb-0_1_1_0", ENV{VE_SERVICE}="ignore"
in the /etc/udev/rules.d/serial-starter.rules

onnimonni, 2024-10-01:
Thanks!

I updated the file and rebooted Cerbo but the delay is the same. When Cerbo restarts it turns on all relays which is not ideal. The related #46 doesn't seem to be happening when the RS485 cable is disconnected.

onnimonni, 2024-10-01:
And just to make sure the 1st USB port is the one on the right upper corner?

[image]

onnimonni, 2024-10-01:
I connected the Relay board to my wifi router and enabled the `TCP Server` option with `Modbus-RTU Over TCP` and it works immediately without any lag now. I wish I can disable the wifi access completely from the board and will try again with another USB->RS485 adapter.

I ordered this one: https://eckstein-shop.de/WaveShare-Industrial-USB-to-RS485-Bidirectional-Converter-EN

This was most likely either the adapter or my loose wiring. The wifi connection also solved #46 for me.

Thanks for the help!

closed 2024-10-01 by onnimonni