Lucifer06/RemoteGPIO #46: All of the relays on the Dingtian board suddenly turned on multiple times

onnimonni, 2024-10-01:
Hey,

I have my setup lying on the table to test it out. 3 times today all of the relays have jumped from the off position to on position.

The relay values do update properly in the Cerbo GX but I'm wondering what is causing this 🤔.

I have the "read relay state" and "read input state" options turned on.

[image]

[image]

Lucifer06, 2024-10-01:
This is more likely linked to issue #45 ...

onnimonni, 2024-10-01:
Seems that this was caused by my `CH41` chip based `USB`->`RS485` adapter. Author recommended to buy one with `FT232R` chip and program it with FTDI tools. See more in #45.