Lucifer06/RemoteGPIO #47: 50-60 second delay between changing relay value and relay actually changing state

pbawesome, 2024-11-12:
I think it might be due to other services trying to use USB0? Is that the top left most USB port if looking at the face of the Cerbo GX? I tried the `/opt/victronenergy/serial-starter/stop-tty.sh ttyUSB0` command but did not notice a difference.

The Dingtian module was already configured for Modbus-RTU on RS485 out of the box, but I did verify it.

[image]
[image]
[image]

pbawesome, 2024-11-12:
The delay is intermittent, sometimes it's nearly instant, sometimes it waits up to 60 seconds. However, the issue I'm having now is that it just randomly switches the relays on its own.

Lucifer06, 2024-11-13:
What is the chipset of your USB to RS485 adapter?
Look at the readme and also at the victron community forum, I have discussed on how to prevent USB port to be prehampted by another process or service. 
I'm 99% sure the issue comes from this problem: you USB0 is not exclusively driven by RemoteGPIO and it makes the communication completly chaotic.

pbawesome, 2024-11-14:
This is the adapter https://www.aliexpress.us/item/3256804592453234.html?spm=a2g0o.order_detail.order_detail_item.3.6d68f19c1nTReQ&gatewayAdapt=glo2usa

I tried following the troubleshooting in your readme but I couldn't seem to make any progress with it.

I did manage to connect the dingtian board to my router and control stuff through UDP which is getting me by for now.

Lucifer06, 2024-11-14:
As you can see on your capture of ps | grep USB0 there is a bunch of processes that are trying to access USB0.
With the specific model of your USB-RS485 adapter, you have a CH340 chip that is generic and can't be reprogrammed with unit serial number for exemple.
So the solution is to modify /etc/udev/rules.d/serial-starter.rules so it ignores the adapter connected to the first physical USB port (when looking at the back of the Cerbo GW, the one at the far right.
So go edit /etc/udev/rules.d/serial-starter.rules and uncomment the following line:
ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_PATH_TAG}=="platform-1c14400_usb-usb-0_1_1_0", ENV{VE_SERVICE}="ignore"
This is normally line #8 

Then reboot the Cerbo and it will solve your problem.

The better alternative is to go with a FT232 chipset, reprogram it to be identified as RemoteGPIO, and then it can be connected on any USB port, including after USB hub.
See instructions here: 
https://communityarchive.victronenergy.com/questions/245584/remotegpio-ex-venus-rgpio-now-available-with-setup.html

onnimonni, 2024-11-15:
I'm currently using the relay over wifi and it also turns all of the relays on every now and then.

I doubt if it's just the usb issue 🤔

Lucifer06, 2024-11-15:
I have also recently see the same. This is related with a bug in the read relay status. Try disable.this option until I fix this. This option is useful to keep the Cerbo displaying the correct current status in case the relay was controlled from another process or.znother server. For ex.if.you use embedded Relay GUI to toggle a relay relays.

onnimonni, 2024-11-18:
I have now been running the Relay board with that setting for 1 hour and I don't anymore see this issue 💯

Thanks ❤️ 

[image]

closed 2024-12-27 by Lucifer06