Lucifer06/RemoteGPIO #51: Add support for the new Switch pane, introduced in guiv2 in v3.60

mpvader, 2025-06-09:
Hey! Making the relays & other outputs if this project available via guiv2 is now possible. API doc is here: https://github.com/victronenergy/venus/wiki/dbus#switch.

[image]
[image]
[image]

Lucifer06, 2025-06-11:
@mpvader as I don't have access to a GX IO-Extender 150, could you provide a dbus-spy screenshot of what looks like the com.victronenergy.switch tree structure? That would help a lot!
Cheers

mpvader, 2025-06-13:
Hey Lucian, can you send me an email with your address? Include the phone number.

Then I’ll send you a GX IO Extender sample.

Lucifer06, 2025-06-14:
> Hey Lucian, can you send me an email with your address? Include the phone number.
> 
> Then I’ll send you a GX IO Extender sample.

That would be cool, thx!
Well it seems GitHub doesn't allow private messages nor the new Victron Community forum.
Could you send me an email to guiot.frederic @ gmail dot com ? And I will reply with my contact details.
Thx in advance,
FreD.

closed 2025-07-03 by Lucifer06

hussl1, 2025-07-03:
@Lucifer06 I already have a Cerbo GX and a Dingtian 8ch-board installed and working in an RV and am ready to mess around with. How can I be of help?

mpvader, 2025-07-21:
Hi @Lucifer06 how is this going?

Lucifer06, 2025-07-22:
I'm making good progress with The Dingtian relays showing up in the new GUI.
At the same time I'm focusing on handling MQTT as the communication driver. It looks much more CPU lightweight.
This is promising.
Stay tuned....

Lucifer06, 2025-07-23:
@drtinaz looks like you are not using mosquito_pub and mosquito_sub. Are you using a MQTT library, which one?
I think using a MQTT library would be preferable, but I was struggling to add  paho-mqtt in Venus OS so I went with the easier route for now...

Lucifer06, 2025-07-23:
I have the DigitalInputs working, but only in polling mode. I'm after offering the digital inputs triggering interrupt (poll mode).

drtinaz, 2025-07-23:
> @drtinaz looks like you are not using mosquito_pub and mosquito_sub. Are you using a MQTT library, which one?
> I think using a MQTT library would be preferable, but I was struggling to add  paho-mqtt in Venus OS so I went with the easier route for now...

I am using paho-mqtt. It took a bit of trial and error to get it working well but it is working quite well. I was fooling around with the digital inputs a little bit. I got as far as creating the dbus and appearing in the device list. I need to do a little work on the inversion selection and callback to the current state. Other than the inversion portion it works well.

Lucifer06, 2025-07-23:
Could you drive me on how you got pho-mqtt installed?
For the DigitalInputs, I'm getting them working by changing the polling mode from epoll to poll mode. This is done by updating /opt/victronenergy/dbus-digitalinputs/start-digitalinputs.sh:
```
#! /bin/sh

set --

if [ -e /dev/gpio/digital_input_1 ]; then
    set -- /dev/gpio/digital_input_*
fi

if [ -d /run/io-ext ]; then
    for f in $(find /run/io-ext -name pins.conf); do
        set -- "$@" --conf $f --poll poll
    done
fi

if [ -z "$*" ]; then
    svc -d .
    exit 0
fi
       
exec $(dirname $0)/dbus_digitalinputs.py "$@"
```

drtinaz, 2025-07-23:
I didn't have to install paho-mqtt i just call it in the script <import paho.mqtt.client as mqtt>

drtinaz, 2025-07-23:
For the digital inputs i just installed them directly as a dbus service. The main driver handles all the dbus services for the virtual devices including callbacks from the gui. One driver controls the switches, temp sensors, tank sensors, virtual battery, and the digital inputs. I have dingtian modules being auto discovered next I'm going to work on shelly devices auto discovery. I still need to add the digital inputs to the auto discovery for the dingtians nut that will have to come later. I'm working on streamlining the setup script at the moment.

Lucifer06, 2025-07-23:
I'm surprised you can manage the DigitalInputs directly as bus service. If not handled by /opt/victronenergy/dbus-digitalinputs/start-digitalinputs.sh  I thought you then missing to see and configure their behaviour from the GUI settings menu.

Do you see them in the menu and you can select a type of input ?

drtinaz, 2025-07-23:
> I'm surprised you can manage the DigitalInputs directly as bus service. If not handled by /opt/victronenergy/dbus-digitalinputs/start-digitalinputs.sh  I thought you then missing to see and configure their behaviour from the GUI settings menu.
> 
> Do you see them in the menu and you can select a type of input ? 
> (just trying to figure out if I lost my entire day trying to compile the kernel for generating my own module to handle the GPIO...)
> 

I don't see them in the settings/digital io menu, but i do see them in the device list and can configure them there. They are handled by my device driver, not the venus digital inputs service.

drtinaz, 2025-07-24:
> Like I said, I'm after making the digital inputs working in efficient way aka epoll method.
> I'm happy to claim I have nail-down my last issues with my kernel module that is mandatory for such behaviors.
> Look at how many Inputs I have at the moment ;-)
> I have to put a limit in the driver, currently 64x DI. I will monitor the performance and cpu usage in real life and will eventually adjust. Next is to merge both my switch support and input support in the same code and release the first beta.
> 
> [image]

Yes we have each taken a different approach. I started by mostly wanting to provide a way for users to be able to add virtual devices without the need for large os or node red, so the digital inputs were an after thought. I have completed the code and am doing some real life testing before i release anything. I have auto discovery of dingtian modules (with support for digital inputs) and shelly switches and plugs. External temp sensors, tank sensors, and virtual battery. I currently have 5 relay modules including digital inputs running, 5 temp sensors, 5 tank sensors, and an external battery. Cpu usage is steady right around 4%. That's just a fraction of the cpu usage required to run even the most basic node red environment.

Lucifer06, 2025-07-24:
Indeed, this is the great benefits of using MQTT.
I have now combined both DigitalInputs and Relay drivers into one.
Next step is to externalize the dependency with Dingtian, so I can add some other brands. Ideally, I would also like to implement Modbus as an alternative for communication (as before and with the CPU load issue). There is still a lot on my plate, but I think I should be able to release a beta version soon

hussl1, 2025-08-01:
Version 4.0 seems to be available. I gave it a try. So far the interface has not changed. Are you still in the makes of implementing? Guimod still is a dependency in setup helper. What files need to be set up tp get it going? What settings need to be done in the relay box?

hussl1, 2025-08-01:
I installed the beta and it works like a charme. Thank you for your work!
Relay control works
Changes in the relay state through mechanical buttons connected to inputs also work fine.

Notes:

In SetupHelper make sure to type Beta and not beta...
I had to delete the existing files from the mainbranch manually, after that SetupHelper downloaded the correct files.
Possible reason: Versions in main and beta are identical.

When i reboot the Cerbo, the switch pane disappears. I have to deinstall and reinstall RemoteGPIO, than it is working again.
Possible reason: All my DC-Consumers but the Cerbo are controlled by a relay which is controlled by one of the Cerbos internal relays via NodeRed. Therfore my Dingtian relay box turns on after the Cerbo is booted.

In the switch pane, 3 Relayboxes are detected, although i only have one. One of them works and it works fine!

One is labled with Setup, my guess is that you re using this one for development and it is therefore a "beta-feature"

The other one did not reapear after i deactivated Modbus on the Dingtian.